bitbank2/BitBang_I2C #5: Slave functionality

oomek, 2020-05-06:
Would it be difficult to extend your library by adding slave functionality? I need to make an i2c injector that sits between a non avr master and a slave. Tried Wire library but it locks up constantly when I use both wire ports in master/slave configuration

bitbank2, 2020-05-06:
It's a good idea, but I haven't had need for it (yet). All it would really require is attaching a pin interrupt to the clock line and adding a little bit of additional logic. Yes, it could be used as a middle-man for an I2C sniffer too. Are you able to write the missing part? I would be happy to merge a PR.

oomek, 2020-05-06:
Let me go through your sources first. I just found your repo.

oomek, 2020-05-06:
Damn, those ifdefs and case sensitive function names duplicates don't help at all :)