bitbank2/BitBang_I2C #6: No activity on Attiny85

oomek, 2020-05-06:
I've started experimenting with your library today, but so far I haven't managed to get any activity on the data lines, even as simple code as this does not produce any output:

```c++
#include <Arduino.h>
#include <BitBang_I2C.h>

BBI2C bbi2c;
uint8_t test[4] = {1,2,3,4};

void setup()
{
	memset(&bbi2c, 0, sizeof(bbi2c));
	bbi2c.bWire = 0;
	bbi2c.iSDA = PB0;
	bbi2c.iSCL = PB2;
	I2CInit(&bbi2c, 400000L);
	delay(100);
}

void loop()
{
	// This works
	// pinMode(PB0, OUTPUT);
	// pinMode(PB2, INPUT);
	// delay(100);
	// pinMode(PB0, INPUT);
	// pinMode(PB2, OUTPUT);

	// This does not
	I2CWrite(&bbi2c, 0x46, test, 4);
	delay(1000);
}
```

Am I doing something wrong?

bitbank2, 2020-05-06:
As it says in the readme, you need to specify the pin numbers differently for AVR to use the fast code.
PB0 should be passed as 0xB0
PB2 should be passed as 0xB2

closed 2020-05-06 by bitbank2

oomek, 2020-05-06:
Sorry, I must have missed that. The waveform looks nice, although the PWM is very wide in low speeds. I've had to set `I2CInit(&bbi2c, 1600000L);` to get 400kHz on Attiny85 btw.

bitbank2, 2020-05-06:
I wasn't accurate with the timing because this library was really aimed at pushing the SSD1306 to go as fast as possible. Most I2C devices are pretty forgiving with the waveform shape and timing.