bitbank2/BitBang_I2C #7: Having trouble

GreyArea1966, 2020-05-19:
This used to work with my display but now seems not to. Same display works with the fullsize driver, so I assume it's just something I'm doing wrong.

Changed pins to 4 and 5 for SDA/SCL as I'm on an Arduino Uno, but just getting blank screen.

bitbank2, 2020-05-19:
A4 != 4. You can specify the pins as port name + bit. For example - PORTB, bit 1 would be 0xb1. Pins A4 and A5 on the Uno can be specified as A4/A5 (they're macros), but if you just use a number, they are pins 18/19.

closed 2020-05-19 by bitbank2

bitbank2, 2020-05-19:
P.S., if you've connected your display to the A4/A5 lines on the Uno, set the pin numbers to -1 and enable hardware I2C.

GreyArea1966, 2020-05-19:
Working now, the -1 and enabling hardware trick worked, many thanks!