bitbank2/BitBang_I2C #10: Detection fails on Chinese AT328P clone nano board

marcdraco, 2020-08-28:
Hi Larry, 

I've come up against a roadblock trying to use your library - neither the BitBang nor the MultiBang libraries work with my hardware and I'm unable to understand why. 

The board is a "nano" clone that uses the "old bootloader" which likely means more to you than it does me. I've got a box of little clones for a couple of projects but a couple of them need displays hence...

I've uploaded several example sketches to the board to ensure everything works (I got caught out early on by some early pictures where the SDA and SCK lines were listed on the D4 and D5 ports... so that was horrible). However, I've even managed to get the Adafruit SSD 1306 demo working like a beaut so I know the base wiring is correct.

As supplied the error comes out as:

```
12:01:23.521 -> Stating I2C Scan
12:01:23.566 -> 0 device(s) found
```

But when I switch the defines:

```
#define SDA_PIN -1
#define SCL_PIN -1
```
The error changes:

```
12:07:23.519 -> Starting I2C Scan
12:07:23.566 -> I2C pins are not correct or the bus is being pulled low by a bad device; unable to run scan
```

Utterly baffled I'll admit - and happy to help tracking this down. In fact, I'm happy to buy you a beer if I can get this working as I really need "multi-bang" or I'll be stuck using one board per display and that's just ... ugh. A multiplexer is an option but for two boards/displays that seems OTT, particularly as space is very much at a premium.

Marc

bitbank2, 2020-08-28:
I'll help if I can. Please provide the sketch that you're using, which wires you're trying and at least a photo/pinout of the board you bought.

marcdraco, 2020-08-28:
Thanks Larry. The demo WIRE library sketches from Arduino work fine which is what has me stumped. 

I'm using a nano clone (pic attached) with the I2C connected to the listed pins - that caused me some headaches as there are still a LOT of images that list the digital pins as the I2C rather than the analog ones. This is the board and you can see the wiring - 3.3v on red, GND on black, SCK on white and SDA on broon (sorry, in joke). 

[image]

This for example which is a slightly pared down example from Oliver Kraus to show off his pretty impressive U8G library works without issue:

```
/*

  GraphicsTest.ino

  Universal 8bit Graphics Library (https://github.com/olikraus/u8g2/)

  Copyright (c) 2016, olikraus@gmail.com
  All rights reserved.
*/

#include <Arduino.h>
#include <U8g2lib.h>
#include <Wire.h>

U8G2_SSD1306_128X32_UNIVISION_2_HW_I2C u8g2(U8G2_R0, 5, 4);

void u8g2_prepare(void) {
  u8g2.setFont(u8g2_font_6x10_tf);
  u8g2.setFontRefHeightExtendedText();
  u8g2.setDrawColor(1);
  u8g2.setFontPosTop();
  u8g2.setFontDirection(0);
}

void u8g2_box_frame(uint8_t a) {
  u8g2.drawStr( 0, 0, "drawBox");
  u8g2.drawBox(5,10,20,10);
  u8g2.drawBox(10+a,15,30,7);
  u8g2.drawStr( 0, 30, "drawFrame");
  u8g2.drawFrame(5,10+30,20,10);
  u8g2.drawFrame(10+a,15+30,30,7);
}

void u8g2_disc_circle(uint8_t a) {
  u8g2.drawStr( 0, 0, "drawDisc");
  u8g2.drawDisc(10,18,9);
  u8g2.drawDisc(24+a,16,7);
  u8g2.drawStr( 0, 30, "drawCircle");
  u8g2.drawCircle(10,18+30,9);
  u8g2.drawCircle(24+a,16+30,7);
}

void u8g2_r_frame(uint8_t a) {
  u8g2.drawStr( 0, 0, "drawRFrame/Box");
  u8g2.drawRFrame(5, 10,40,30, a+1);
  u8g2.drawRBox(50, 10,25,40, a+1);
}

void u8g2_string(uint8_t a) {
  u8g2.setFontDirection(0);
  u8g2.drawStr(30+a,31, " 0");
  u8g2.setFontDirection(1);
  u8g2.drawStr(30,31+a, " 90");
  u8g2.setFontDirection(2);
  u8g2.drawStr(30-a,31, " 180");
  u8g2.setFontDirection(3);
  u8g2.drawStr(30,31-a, " 270");
}

void u8g2_line(uint8_t a) {
  u8g2.drawStr( 0, 0, "drawLine");
[... 145 lines cut ...]
    case 10: u8g2_bitmap_modes(1); break;
    case 11: u8g2_bitmap_overlay(draw_state&7); break;
  }
}

void setup(void) {
  u8g2.begin();
}

void loop(void) {
  // picture loop  
  u8g2.clearBuffer();
  draw();
  u8g2.sendBuffer();
  
  // increase the state
  draw_state++;
  if ( draw_state >= 12*8 )
    draw_state = 0;

  // deley between each page
  delay(100);

}
```

bitbank2, 2020-08-28:
Thanks for the photo and the wiring, but I need to see the code you wrote which uses my library to see if you've made an error.

marcdraco, 2020-08-28:
Literally just this. The only change I made (so far) is to switch to the internal WIRE lib as instructed by switching the definitions to "-1". I'm not 100% familiar with the numbering layout so I've used the old DIP counting starting from 1 (top left) and counting round and back up. By all means correct any mistake/assumption I've made as I'm in a bit of a fix without your library!

```
//
// I2C Detector - scan and identify devices on an I2C bus using the BitBang_I2C library
// 
// The purpose of this code is to provide a sample sketch which can serve
// to detect not only the addresses of I2C devices, but what type of device each one is.
// So far, I've added the 25 devices I've personally used or found to be reliably detected
// based on their register contents. I encourage people to do pull requests to add support
// for more devices to make this code have wider appeal.

// There are plenty of I2C devices which appear at fixed addresses, yet don't have unique
// "Who_Am_I" registers or other data to reliably identify them. It's certainly possible to
// write code which initializes these devices and tries to verify their identity. This can
// potentially damage them and would necessarily increase the code size. I would like to keep
// the size of this code small enough so that it can be included in many microcontroller 
// projects where code space is scarce.

// Copyright (c) 2019 BitBank Software, Inc.
// Written by Larry Bank
// email: bitbank@pobox.com
// Project started 25/02/2019
//
// This program is free software: you can redistribute it and/or modify
// it under the terms of the GNU General Public License as published by
// the Free Software Foundation, either version 3 of the License, or
// (at your option) any later version.
//
// This program is distributed in the hope that it will be useful,
// but WITHOUT ANY WARRANTY; without even the implied warranty of
// MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
// GNU General Public License for more details.
//
// You should have received a copy of the GNU General Public License
// along with this program.  If not, see <http://www.gnu.org/licenses/>.
//
// Uses my Bit Bang I2C library. You can find it here:
// https://github.com/bitbank2/BitBang_I2C
#include <BitBang_I2C.h>

// Arbitrary pins I used for testing with an ATmega328p
// Define as -1, -1 to use the Wire library over the default I2C interface
//#define SDA_PIN -1
//#define SCL_PIN -1
// M5Stack Atom Grove connector pin assignments
//#define SDA_PIN 32 
//#define SCL_PIN 26
// M5Stack Atom internal I2C connected to the IMU
#define SDA_PIN -1
#define SCL_PIN -1
//
// If you don't need the explicit device names displayed, disable this code by
// commenting out the next line
//
#define SHOW_NAME
#ifdef SHOW_NAME
const char *szNames[]  = {"Unknown","SSD1306","SH1106","VL53L0X","BMP180", "BMP280","BME280",
                "MPU-60x0", "MPU-9250", "MCP9808","LSM6DS3", "ADXL345", "ADS1115","MAX44009",
                "MAG3110", "CCS811", "HTS221", "LPS25H", "LSM9DS1","LM8330", "DS3231", "LIS3DH",
                "LIS3DSH","INA219","SHT3X","HDC1080","MPU6886"};
#endif

BBI2C bbi2c;

void setup() {
  Serial.begin(115200);
  memset(&bbi2c, 0, sizeof(bbi2c));
  bbi2c.bWire = 0; // use bit bang, not wire library
  bbi2c.iSDA = SDA_PIN;
  bbi2c.iSCL = SCL_PIN;
  I2CInit(&bbi2c, 100000L);
  delay(100); // allow devices to power up
}

void loop() {
uint8_t map[16];
uint8_t i;
int iDevice, iCount;

  Serial.println("Starting I2C Scan");
  I2CScan(&bbi2c, map); // get bitmap of connected I2C devices
  if (map[0] == 0xfe) // something is wrong with the I2C bus
  {
    Serial.println("I2C pins are not correct or the bus is being pulled low by a bad device; unable to run scan");
  }
  else
  {
    iCount = 0;
    for (i=1; i<128; i++) // skip address 0 (general call address) since more than 1 device can respond
    {
      if (map[i>>3] & (1 << (i & 7))) // device found
      {
        iCount++;
        Serial.print("Device found at 0x");
        Serial.print(i, HEX);
        iDevice = I2CDiscoverDevice(&bbi2c, i);
        Serial.print(", type = ");
  #ifdef SHOW_NAME
        Serial.println(szNames[iDevice]); // show the device name as a string
  #else
        Serial.println(iDevice); // show the device name as the enum index
  #endif
      }
    } // for i
    Serial.print(iCount, DEC);
    Serial.println(" device(s) found");
  }
  delay(5000);
}
```

bitbank2, 2020-08-28:
You're not telling it which pins to use. If you want to use the default hardware I2C, then you can set the pin numbers to -1. For bit banging, you need to specific the pins as the Arduino GPIO pin numbers. For the Nano, A4 = A4 or 18, A5 = A5 or 19. For faster access you can also use a pin numbering I came up with to specific the ATMega port numbers. A4 = 0xC4, A5 = 0xC5. Those are pins on PORTC. The point of my library is that you can use ANY pins, not just A4/A5. Does that make sense?

closed 2020-08-28 by bitbank2

marcdraco, 2020-08-28:
I got the idea of the library which is why I was desperate to use it. I need to run two or even three displays over i2C even if they don't need particularly fast data. Anyhoo...

Inputting your numbering works beautifully (*and thank you, some beer coming your way shortly). Perhaps, if you'd be kind enough, you can explain how the this *****ing numbering works. This is one of the reference images I used (apart from manual counting which fails for the same reason). You'll note also that this one has that error on the SDA/SCK mapping - suggesting that they are on the digital inputs not the analog ones. 

My long distant background in very basic TTL/MOS electronics taught me to count the numbers as on this diagram.

[image]

So A4/A5 would map to 23 and 24 if that makes sense - I'm baffled as to how this comes out as 18/19 - though there's no question that it works perfectly! I haven't tried anything else that "numbers" pins with such distinction so I might have missed the memo that explains how to count the pins vs. the "DIL package" numbering.

"-1" (per your comment in the code)  doesn't seem to map to 18/19 though. Quite why anyone would want to use WIRE when you have this is... over my head.

I'll admit that I don't understand the hex coding for the ATMega. I'm quite comfortable with hex (I've hand coded in Z80 assembler back in the day - it's not much fun!) but the Mega is simply a black box to me if you get my drift.

If you can point me to a better reference so I won't screw this up again, I'd really appreciate it. Particularly understanding how those pin numbers map to actual numbers would be a great help. As it is I'm still unable to infer how those work - which is nuts on a system designed to teach electronics. I'm just using it as a means to an end rather than the end itself.

Thank you for a great library Larry, I can't really tell you how grateful I am for your help.

bitbank2, 2020-08-28:
[image]
In this image, the purple pins are the Arduino pin numbers to use with pinMode, digitalWrite, etc.
I came up with the hex numbering as a way of directly specifying the ATMega port and bit. Port C, bit 4 is 0xC4 - convenient, no?

marcdraco, 2020-08-28:
Ah, now THAT makes far more sense. I've been trying to navigate a desert without a map, compass or any clue really. Your numbering system is great - it would make more sense if I knew what port C was on the ATMega but that's another story.  

The problem with being a noob descending from the days of 74- series TTL and 40- series CMOS is I'm only used to referencing pins per DIL and similar schemes start at one (were 31 is here) and count round. I assume this disparity is something to do with keeping the board simple and only two layers? Where I trained we'd be required to run the various inputs/outputs for something more, well, obvious. That makes it more complex because we have to jumper the tracks and so on, but it's certainly dooable at the expense of a more expensive board.

Is there any call for a pin translation? Indeed perhaps there is a library that does it that I'm unaware of yet (I'm still plodding through the list). What I mean is something (and I presume it could be done with defines) that says "pin 19" - ah, that's DIL pin 24. (At least is would be on a nano...) In fact this is precisely how I cocked up with your lib - I was using pin 24 & 25 rather than 18 & 19 because 23 & 24 are marked as A4 and A5. It wasn't your code that was wrong (d'uh me!), it was my interpretation of the pin numbers.

Hopefully this conversation will help some other poor sap who falls into the same trap!

bitbank2, 2020-08-28:
It certainly is confusing. All of my libraries for Arduino use the Arduino pin numbers. These numbers go through a translation table in the core libraries to turn them into GPIO port and bit numbers. The ATmega series has 8-bit GPIO ports A-G (depending on the chip). The Arduino pin functions (pinMode, digitalWrite, digitalRead) are somewhat slow because they translate the pin numbers, compare for valid values and disable interrupts each call. When doing bit-banging this slows things down quite a bit. I came up with the hex number system to allow a user to specify the hardware port info on any ATmega to avoid the slowness of Arduino's generalized code. The available ports and pins is different for each ATmega chip, so I can't make any assumptions.