bitbank2/BitBang_I2C #12: BME680 recognition fails

cyberman54, 2020-10-13:
I have a device with working BME680. I integrated the example code for i2c discovery in my application. It seems it works, i2c devices are found, and also the BME680 ist found. But it's not recognized. Looks like reading the chip ID register 0xd0 fails for some reason. But the BME680 deliveres valid values by i2c to my application.

[I][i2c.cpp:48] i2c_scan(): Starting I2C bus scan...
[I][i2c.cpp:79] i2c_scan(): Device found at 0x3C, type = SSD1306
[I][i2c.cpp:79] i2c_scan(): Device found at 0x76, type = Unknown
[I][i2c.cpp:82] i2c_scan(): 2 I2C device(s) found

bitbank2, 2020-10-13:
Can you read the value returned from register 0xd0? I can add it as another variation to the code. Mine returns 0x61.

cyberman54, 2020-10-14:
This was a false positive, sorry.
Turned out, platformio had still 2.1.2 installed. With 2.1.3 BME680 is recognized.

closed 2020-10-14 by cyberman54