bitbank2/BitBang_I2C #21: Raspberry Pi: how to initialize the lib for default hardware i2c-0/1 and how to specify OLED type and i2cAddr directly?

dsyleixa, 2021-04-18:
hi,
for the Raspberry Pi (OS Stretch): 
how to initialize the lib for default hardware i2c-0  and/or i2c-1 (pin numbers? which numbering (BCM, physical)?  or how else?)
how to specify an arbitrary OLED type (e.g. SSSD1306)
and how to specify  i2cAddr directly (no autodetect, e.g. 0x3c)?

bitbank2, 2021-04-18:
See the pigpio documentation for how pin numbers are mapped (that's what's used within BitBang_I2C on the RPI).
The OLED type is specified with a constant defined in ss_oled.h
The OLED address is auto-detected (see line 792 in ss_oled.c)
The I2C clock speed of the RPI can be set to a max of 400K see here: https://www.raspberrypi-spy.co.uk/2018/02/change-raspberry-pi-i2c-bus-speed/

closed 2021-04-18 by bitbank2

dsyleixa, 2021-04-18:
ok, pin numbers then probably by BCM-numbering, so for i2c-1 probably SDA=2 , SCL=3 
DEVICE_SSD1306 is the 1st one mentioned in the 2nd enum line, so perhaps = 1?

But I would like no i2c-auto-detect but explicitely set the i2cAddr explicitely :  is that not possible?

bitbank2, 2021-04-18:
So set the address. If it's -1, then it auto-detects.
The pin numbers are only needed if you want to bit-bang the I2C protocol. If you want to use the hardware I2C, then you don't need to provide pin numbers.

dsyleixa, 2021-04-18:
sorry, I am quite a beginner, it's completely confusing me...
how would the command line look like to initialize the lib for 
standard i2c-1 
SSD1306
addr 0x3c

bitbank2, 2021-04-18:
Look at the examples in ss_oled. The code on the RPI will be basically identical.