bitbank2/BitBang_I2C #23: Warning! Could not find the package with 'bitbank2/BitBang_I2C @ ^2.1.3' requirements for your system 'windows_amd64'

wally2511, 2021-06-08:
Hi

Just compiling the latest version of PaxCounter and it is getting hung up on this library. I can't seem to get it to install!
I'm running Visual Studio Code and using Platformio.

**Warning! Could not find the package with 'bitbank2/BitBang_I2C @ ^2.1.3' requirements for your system 'windows_amd64'**

`LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ deep, Compatibility ~ soft
Library Manager: Installing bitbank2/OneBitDisplay @ ^1.10.0
Warning! Could not find the package with 'bitbank2/OneBitDisplay @ ^1.10.0' requirements for your system 'windows_amd64'
Library Manager: Installing bitbank2/BitBang_I2C @ ^2.1.3
Warning! Could not find the package with 'bitbank2/BitBang_I2C @ ^2.1.3' requirements for your system 'windows_amd64'
Found 50 compatible libraries
Scanning dependencies...
Dependency Graph`

bitbank2, 2021-06-08:
Maybe this has something to do with the internet outage today, but otherwise I haven't seen any problems like this.

closed 2021-06-08 by bitbank2

cyberman54, 2021-07-20:
This problem still persists.
Platformio library manager suggests this declaration in platformio.ini, but still results in above error.

```
lib_deps =
  # RECOMMENDED
  # Accept new functionality in a backwards compatible manner and patches
  laurencebank@gmail.com/BitBang_I2C @ ^2.1.3
```