bitbank2/BitBang_I2C #27: Wire lib emulation

hydrastarmaster, 2021-08-29:
At the moment, this great piece of sw seems to be a stand-alone lab tool. To encourage its use in projects, it would be very useful to add a compatibility layer with the Wire library, i.e. add some dummy functions that emulate the ones from Wire. In this way, we will just substitute the #include "Wire" with the #include "BitBang_I2C". It's surely better than rewrite the i/o for every single library. Do you think it would take a lot of effort?

hydrastarmaster, 2021-08-29:
Message from: Adafruit
we'd accept a bitbang i2c implementation in Adafruit BusIO so it's abstracted out, then it could be added easily to any sensor. please submit a PR when you can!

bitbank2, 2021-08-30:
I wrote this library for my own projects and then decided to publish it because I thought it would be useful for the community. It already includes the ability to mix and match hardware I2C with bitbang on multiple buses simultaneously. I don't plan on changing the API to match the Wire library because there is already an Arduino library which does that. This library is meant to be used when people have unique needs such as managing multiple devices - some connected to the hardware I2C of the MCU and need additional devices (maybe with conflicting addresses) connected on spare GPIO pins.

closed 2021-08-30 by bitbank2

unitware, 2021-11-15:
I am also searching for a wire.h compatible bitbang library. You say that it exists but I have not been able to find it, would it be possible to share a link?

bitbank2, 2021-11-15:
I haven't used it, but this is one of several libraries that mimics the Wire API:
https://github.com/Testato/SoftwareWire