bitbank2/BitBang_I2C #30: Arduino Example No Longer Works

janbbeck, 2022-06-28:
The Arduino example I2C_Detector calls I2CDiscoverDevice with two parameters.
However the current version of BitBang_I2C asks for 3, causing the compile to fail.

Going to version 1.x in the LibaryManager lets BitBang work again.

FutureTEK99, 2022-11-11:
Arduino will call setup() and then call loop() over and over in an infinite loop.
also:
 Compilation error: too few arguments to function 'int I2CDiscoverDevice(BBI2C*, uint8_t, uint32_t*)'

yosory, 2025-03-11:
I think I ran into the same problem today. In the main.c where I2CDiscoverDevice is called, I added NULL as third argument `iDevice = I2CDiscoverDevice(&bbi2c, i, NULL);`

bitbank2, 2025-03-11:
I updated the function to use uint32_t and match the example. It is pushed to main, but not in an official release yet.

closed 2025-03-11 by bitbank2